copl-labomc/fabconcept #12: Change the Cabestan motor

alexmomo1212, 2024-05-10:
We need a new motor with lower torque and higher speed. Preferably, it should be from the same manufacturer so the size and shape is the same as the old one.

cniallen, 2024-05-14:
As the purchase order form is getting re-formatted, let @joycopl and me know if you select the model to order before I publish the new form to use. We'll order for you with the temporary one if needed.

joycopl, 2024-05-14:
The speed of the motor should be much higher than we have now. Check the similar motor with as much as possible higher speed. We will control the speed using potentiometer according to our requirements.

alexmomo1212, 2024-05-28:
The new motor has arrived all we need is to change the motor holder (Issue #31)

closed 2024-05-28 by alexmomo1212