copl-labomc/fabconcept #9: Circuit schematic

alexmomo1212, 2024-05-10:
Draw a circuit for the fiber tower's electronics.

It can be done with Fritzing, TinkerCAD or Altium.

cniallen, 2024-05-14:
We a strong preference for testing out Fritzing for educational purposes please.

cniallen, 2024-08-04:
I like the way the documentation is going, but I feel it's missing a little something to tie it together on the home-landing page. It's main theme should be "Welcome new fiber tower developper!" and, after redirecting a user to the "User guide" page with a link, give to the newcomer an overview that helps her or him get started in the advent everybody is too occupied to help right away. I guess it should be along the lines of what @grudule explained to @alexmomo1212 at the beginning of the summer. 

I think it needs a section on how our workflow to work with github, small new features branches to be merged in the development branch first, test an executable, pull request to the main brach with code review and /fast-forward merge (assuming with fix he missing token!), etc. I think one of the first things @grudule did in the lab was a diagram for this workflow isn't it? I think it's worth putting it in that section.

I would also definitely bring the functional diagram on the homepage, is the original of that diagram in the repo somewhere btw? I think it needs a little updating and very much replacing "laser sensor" by "diameter sensor" please. 

I think a section giving some info on the relations between the code and the circuit diagram would also help, since otherwise one as to dig in the code to figure out the ins and outs on the Arduino and drivers. Actually, those pins do not figure as some variables or arguments somewhere in the GUI_doc?

Thanks in advance for these improvements!